ICOMP-UNC/Lenox-Legends #5: Configuracion del entorno de desarrollo.

franivan98, 2024-11-11:
Tenemos que configurar y preparar el Visual Studio Code para poder trabajar con la STM32 y con Free RTOS.

Lista de Actividades:

- [ ] Instalar la extension "PlatformIO".
- [ ] Descargar e incluir la libreria "rtos" en el dicrectorio del proyecto. (Lo pasaron por el grupo de Discord)
- [ ] Compilar un codigo de prueba.

franivan98, 2024-11-11:
Terminado por mi parte

Lenox30, 2024-11-12:
Se instalo PlatformIO, se descargo la librearia rtos, se compila bien el programa sin embargo no puedo ver las librerias de la STM32 creo que por algun problema de visual studios

Mauricho, 2024-11-12:
Se ha logrado instalar Visual Studio Code junto con la extensión PlatformIO para hacer parpadear el led P13 del puerto C de la STM32.

Lenox30, 2024-11-12:
Dado que mi Visual Studio de Windows se rompio totalmente, decidi pasar a linux para probar desde 0 y se ha logrado instalar, compilar y ver todas las librerias tanto de RTOS como de libopencm3

franivan98, 2024-11-12:
Listo, lo damos por cerrado

closed 2024-11-12 by franivan98