ICOMP-UNC/Lenox-Legends #8: Diagrama Circuital

franivan98, 2024-11-11:
Definir las conexiones de la STM 32. 
Lista de componentes y pines a usar:

Mauricho, 2024-11-15:
Versión 1 del esquemático.
[image]

Mauricho, 2024-11-15:
Versión 2 del esquematico.
[image]

Mauricho, 2024-11-18:
Se realizan modificaciones al diagrama circuital. Se actualiza según las últimas implentaciones.
[image]

franivan98, 2024-11-19:
Agregar Pines para motores:
PA6: Motor 1
PA7: Motor 2

Mauricho, 2024-11-20:
Se actualiza el esquematico y el diagrama circuital de acuerdo a los realizado.
[image]

closed 2024-11-20 by franivan98